OpenRoberta/robertalab #194: Welcome screen links to non-existant building instructions

murraycu, 2016-08-07:
The welcome screen has a "building instructions" link (actually some JavaScript thing, I think) which I guess is meant to show a suitable lego model to build for use with the Open Roberta Lab, but it takes us to this (non-existant) TODO page:
https://lab.open-roberta.org/TODO

murraycu, 2016-09-11:
There's some status about this here: https://github.com/OpenRoberta/robertalab/issues/203#issuecomment-240360047

murraycu, 2016-09-11:
If those instructions then require parts from only the EV3 Education or regular EV3 set, the FAQ should be updated to state that. It currently says "In general it does not matter if you got the retail set (from the toy department) or the education set (from LEGO Education).":
https://mp-devel.iais.fraunhofer.de/wiki/display/ORInfo/FAQ

I imagine it would be difficult to design one model that works with both sets.

bjost2s, 2016-09-21:
fixed in 3d25941562852a4121061adb8e1806c05bc48479

closed 2016-09-21 by bjost2s

murraycu, 2016-09-22:
Thanks.

But that commit was a massive collection of unrelated stuff. It should have been several smaller commits, with appropriate commit messages, though I guess the blockly update could be one commit. I strongly suggest using "git add -p" and suchlike.

murraycu, 2017-07-10:
I can't now find any EV3 building instructions anywhere. I don't know what commit removed the link, but I don't think that huge commit had anything to do with it. I've filed a new bug #555 about it.