OpenRoberta/robertalab #201: improve the generated code of wait-statements

ensonic, 2016-08-12:
The wait-statements generate quite verbose code right now:

``` python
while True:
  if ( hal.getUltraSonicSensorDistance('4') < 48 ) or hal.isKeyPressed('enter'):
      break
  hal.waitFor(15)
```

It would be nice to use this instead:
``` python
while not ( hal.getUltraSonicSensorDistance('4') < 48 ) or hal.isKeyPressed('enter'):
  hal.waitFor(15)
```

Unfortunately the conditions on connects to the wait-until block are pre-wrapped with 'if' + 'break' somewhere, so that right now the codegen would create:
``` python
while not if hal.getUltraSonicSensorDistance('4') < 48:
  break:
  hal.waitFor(15)
```

I think the waitstatement should contain a simple (compound) statement with the waiting condition.

cvejoski, 2016-08-12:
We can change the code when we have only one conditions statement like you propose in the ticket, but for the following case:
`while ( true ) {
            if ( hal.isPressed(SensorPort.S1) == true ) {
                hal.driveDistance(DriveDirection.FOREWARD, 30, 20);
                break;
            }
            if ( hal.getColorSensorAmbient(SensorPort.S3) < 50 ) {
                hal.drawText("Hallo", 0, 0);
                break;
            }
            hal.waitFor(15);
        }`

this would no work.

ensonic, 2016-08-12:
I am talking about the "wait-until" block. This cannot have children:
[image]

closed 2016-09-21 by cvejoski