OpenRoberta/robertalab #203: update 'track width' in brick configuration to match the default robot model

ensonic, 2016-08-12:
The track width is currently set to 18cm by default. In the default roberta model the track width is 12cm.

bjost2s, 2016-08-14:
The default Open Roberta Model has a track width of 18cm. Unfortunately the link is brocken to the instructions. This should be fixed. See #194

ensonic, 2016-08-15:
Is the model different to what roberta teachers usually built ("Bauanleitung RobertaEV3")? That would be sad.

Thorlogy, 2016-08-17:
The original / official Roberta model isnt perfect for mounting the color sensor. Therefore we decided to design an Open Roberta EV3 model...  the plan is to integrate this model in the updated Roberta-Grundlagenband which will be published end of October.

ensonic, 2016-08-17:
I am happy to test the new instructions. I made a few small tweaks to the current instructions to manage the cables.

closed 2016-09-21 by cvejoski