OpenRoberta/robertalab #218: Stop program

Thorlogy, 2016-08-19:
Program should be stopped via Open Roberta Lab

The start button could be change to stop button. 

This would help to stop programms with endless loops

ensonic, 2016-08-19:
This won't work right now, as we suspend the the push protocol while the program runs. Other than that I would prefer this behaviour too, maybe plan properly and do it after 2.0

bjost2s, 2016-09-27:
only for ev3, the only robot connected via wifi

ensonic, 2017-12-15:
In the long run I'd like to keep to make a change to the protocol to keep it running while the program runs, but it is not a simple as it sounds:
* the program can also be run from the robot, people need to understand that in that case, the will have to stop it from the robot
* when running from the robot, the programm needs to understand that it is not conencted to the robertalab, so that it won't try to send telemetry data
* only robots with a link (wifi, bluetooth, cable?) could potentially support stopping the execution from the UI

rbudde, 2019-06-26:
not feasable at the moment

closed 2019-06-26 by rbudde