OpenRoberta/robertalab #224: Changes in blocks for botnroll

elhele, 2016-08-24:
Action

1.  Move  blocks.
First there was no option for moving only one motor in the library. Now José  is adding new functions. However, I'm not entirely sure that he'll make it till September. In this case two motor function and two speed variables to store non-changed motor speeds will be used. 
- Blocks "motor port on speed" and "set motor speed" are the same for this robot, we need to merge them. 
- Block "motor port on speed for rotation/degree" can not be implemented, because there is no encoder.
- Block "get speed motor port" can not be implemented because there is no encoder.
- Block "stop motor port float/brake" can be implemented after José  creates a function for separate stop of the motors with these modes (not sure that till September). So far it can only be done using a function for motor speed with a speed of the required motor set to 0.

2. Drive blocks.
There are no functions exactly for turning, all blocks are implemented using move motors function.
- Block "drive forwards/backwards speed" is not a problem, there is a function for it in the library.
- Block "stop" is not a problem, there is a function for it in the library (stop, "It cuts the energy to the motors and they rotate freely until they stop"). There is also a function for brake. Perhaps we could add it into block.
- Blocks "drive forwards/backwards speed" and "turn right/left degree" are hard to implement without encoder. If we try to rely on time, it won't work since speed depends on battery. José  was going to fix it, but not clear when excalty. We could change blocks so it would work with time instead of distance.

3. Display blocks work fine, but instead of "in column" and "in row" we have should have  only lcd line number (1 or 2)- there are two functions lcd1 and lcd2.

4. Sounds.
- Block "play frequency duration" works well;
- Blocks "set volume" and "get volume" are reallz hard to implement on Arduino. As I understand, it maz require some hardware changes.

5. Lights. As we discussed, we can use a block similar to EV3, but without colors and with only "on/off" options.

bjost2s, 2016-10-03:
I think encoders are available, so not removing the encoder sensors make sense.

elhele, 2016-10-03:
On some models- the older ones- encoders are available, but, as I understand, they removed the encoders from the newer models. I guess yes, we just could make it possible to remove and include them in the robot configuration.

closed 2016-11-14 by bjost2s