OpenRoberta/robertalab #253: Plugin structure to ease integration of new robots

rbudde, 2016-10-05:
Goal:
if a new robot is integrated, this must be a simple process. Modification of generic code should not be necessary (every violation of this rule needs a good argument).

Todos:
- allow registration of new robots based on a configuration file
- allow robots to add new blocks and give developers the possibility to reuse existing code generators and to extend them

bjost2s, 2016-11-14:
This works very well and makes it much easier to add new systems!

bjost2s, 2016-11-22:
Last thing to do:
separate EV3lejos and EV3dev. Treating the both systems as two different robots plugins will make the workflow for developers and users! much easier to understand.

bjost2s, 2016-11-22:
see #308

closed 2016-11-22 by bjost2s