OpenRoberta/robertalab #281: usb program is not detecting the EV3 on linux

Jaleks, 2016-11-05:
Same as [#168](https://github.com/OpenRoberta/robertalab/issues/168) only with EV3

bjost2s, 2016-11-15:
did you installed the libraries from #168?

Jaleks, 2016-11-15:
yep: 
$ dpkg --status libc6-i386 libxext6:i386 libxrender1:i386 libXtst6:i386 libXi6:i386 libusb-0.1-4:i386 | grep -e Package -e Status -e Version 
Package: libc6-i386
Status: install ok installed
Version: 2.19-18+deb8u6
Package: libxext6
Status: install ok installed
Version: 2:1.3.3-1
Package: libxrender1
Status: install ok installed
Version: 1:0.9.8-1+b1
Package: libxtst6
Status: install ok installed
Version: 2:1.2.2-1+b1
Package: libxi6
Status: install ok installed
Version: 2:1.7.4-1+b2
Package: libusb-0.1-4
Status: install ok installed
Version: 2:0.1.12-25

cvejoski, 2016-12-05:
You have to use also java 32bit version because we use for the NXT the Fantom that works only with 32 bit java

Jaleks, 2016-12-05:
> we use for the NXT 
so also for the EV3 a (somehow?) parallel installation of 32bit java is needed?

cvejoski, 2016-12-05:
Yes, at the moment because of the NXT we need the Roberta USB program to be run with Java 32bit for every robot. 

We work currently to create debian package for installing the USB program on linux. We also work to resolve the requirement of the 32 bit java so in future for a user to be able to use any java version.

closed 2016-12-05 by Jaleks