OpenRoberta/robertalab #283: Syntax Error in a program with nested loops

filipwoz, 2016-11-07:
Please help! This issue is a show stopper for my students.
By starting the simulation of the attached program
[TestLoops.zip](https://github.com/OpenRoberta/robertalab/files/575028/TestLoops.zip)
[TestLoops screenshots.pdf](https://github.com/OpenRoberta/robertalab/files/575038/TestLoops.screenshots.pdf)

the simuation is not started and the error message appears in the log:
[[ERR ]]	[[TIME]] 23 msec: run program 'TestLoops' with configuration 'NXTbasis', then EXCEPTION: SyntaxError: Unexpected token var

Screenshots of the program attached as well.

filipwoz, 2016-11-07:
Additional info: we did that in the NXT mode

cvejoski, 2016-11-22:
fixed with a2ffb3523902f80a608133ab04abe3e4a3cfa7ab

closed 2016-11-22 by cvejoski