OpenRoberta/robertalab #284: Uploading the program without internet connection

filipwoz, 2016-11-07:
We plan to use Roberta in the Open Category of the WRO contest. 
But in the contest sites the Wifi with internet connection is not guaranteed. 
We need a possibility to make changes to our program and upload it to a robot without internet connection.

ensonic, 2016-11-07:
Can you use a local openroberta server on a raspberry pi?
https://github.com/OpenRoberta/robertalab/wiki/Raspberry-Pi-2---3

filipwoz, 2016-11-07:
I thought that this could be a solution. But is it the only option at the moment? Can I also setup a local openroberta server on a laptop?

ensonic, 2016-11-07:
Sure, you can also run the openroberta serer directly on your laptop. You will need to turn your laptop into a wifi AP or use the usb-program and connect the robot via usb.

filipwoz, 2016-11-07:
OK I have to try it out.
And how will the USB-program know that it should communicate with my Laptop and not with the Open Roberta server in Internet?

ensonic, 2016-11-07:
There is a checkbox on the bottom, where you can enter the ip/port for the local server. We unfortunately don't have prebuild packages. Allthough it is straight forward to build. For what platform would you want the pre-built package?

filipwoz, 2016-11-08:
I would like to run it on a Windows laptop - and preferably connect NXTs and EV3s via USB.

bjost2s, 2016-11-13:
We use exactly this setup for faires and demos, where internet is not guaranteed, works fine for windows, mac, linux in combination with the usb program.

filipwoz, 2016-11-14:
Then I would appreciate a step by step setup description similarly like for what you did for Raspi.

bjost2s, 2016-11-15:
Please ask if you have questions, there is a basic setup description for running the server on windows.
Let me know if you would like to write a description ;-)

Otherwise it will take some time

closed 2017-11-25 by bjost2s