OpenRoberta/robertalab #285: EV3: "Drive forward speed ...." does not start the motors continously

filipwoz, 2016-11-10:
I have written the attached program and tried it out on EV3. In simulator the program runs perfectly.
By comming to the "Drive forward speed ...."  command the EV3 robot only moves the wheels very slightly as though the driving was turned on for a moment and then turned off again.
What I expect is that it drives continously untill the black color is detected or an obstacle is found by the distance sensor.
[EV3 Simple Sumo.zip](https://github.com/OpenRoberta/robertalab/files/583408/EV3.Simple.Sumo.zip)

ensonic, 2016-11-10:
I've just tried it out (on ev3dev.org) and it works great. Is this using lejos or ev3dev on the ev3?

filipwoz, 2016-11-10:
I have used lejos. Is the ev3dev.org already fully supported? I thought it is first some prototype version, as the user friendly installation guide is present only for Lejos.

ensonic, 2016-11-10:
ev3dev works, if you try the latest image from oct this year, your need to update the packages once though.
I'd be happy to get more testers for it.

filipwoz, 2016-11-11:
So you mean ev3dev fully supports all Open Roberta blocks?
I did not get you; what do I need to update in the latest ev3dev package? How?
Does open roberta (the version on the web) recognize automatically that I have ev3dev and generate then Python code?

ensonic, 2016-11-11:
Yes, if you have another sd card, download an ev3dev image, write it to the card and boot from it. Setup wifi according to the docs on ev3dev. To use openroberta you can follow the instructions on:
https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/README.md#intro

In addition, when you are logged into the ev3, you will need to update packages one more time (sudo apt-get update, sudo apt-get upgrade). This is because we had a release after the last image was made. A new image is in preparation and once that is out this step won't be needed.

On the openroberta side, it is all auto-detected. ev3dev supports the same blocks and more in the future (e.g. speech output). If an ev3 running ev3dev firmware is connected you automatically get python.

bjost2s, 2016-11-15:
Unfortunately we can not fix this error, because it is caused by a low level method in lejos. Currently there is also no other, similar method available.

filipwoz, 2016-11-15:
If this is not working in lejos then no "real-time" handling can be implemented where the sensors are checked while motors are running. I think it must work in lejos. I will try to write an equivalent program  in pure java/lajos and let you know.

bjost2s, 2016-11-19:
If you want to do real-time programming, please set your main motors to be unregulated in the configuration. You cannot use the drive functions anymore, because they need the regulation, but your can create your own regulation and you got real time behaviour.

bjost2s, 2017-08-26:
Solution in sight! With OpenRoberta/robertalab-ev3lejos#6 we can use the chassis class to fix this issue.

closed 2017-09-29 by cvejoski