OpenRoberta/robertalab #286: Server Error: when in robot configuration wheel diameter with , instead of . is given

filipwoz, 2016-11-10:
When I put in robot configuration wheel diameter 3,6 and start the simulation a server error pop up appears
Better error handling or field content format validation would be good.

bjost2s, 2016-11-15:
cannot reproduce the error
can you make a screenshot of the error message?
btw. the simulation doesn't care about configuration, it uses always the standard configuration.

filipwoz, 2016-11-15:
Steps to reproduce in attachment
[Steps to reproduce.pdf](https://github.com/OpenRoberta/robertalab/files/591418/Steps.to.reproduce.pdf)

bjost2s, 2016-11-21:
Thanks, the error is more or less the message. usually we ignore configuration modifications in the simulation

cvejoski, 2016-11-23:
I was trying to reproduce the error following the instruction from @filipwoz  but everything was fine.

Can you please tell me @filipwoz which browser and OS have you used? After changing the values in the configuration for confirmation did you use the ENTER or TAB key or you just simply switch to another   tab?

bjost2s, 2017-02-01:
@filipwoz please reopen you find a way to reproduce it

closed 2017-02-01 by bjost2s