OpenRoberta/robertalab #287: use webbluetooth to avoid the need for a helper program

ensonic, 2016-11-11:
webbluetooth is shaping up. This would allow us to talk to bluetooth devices directly from the browser (js), without having an extra proxy.

Spec: https://webbluetoothcg.github.io/web-bluetooth/

Here is the current status: 
* http://caniuse.com/#search=webbluetooth
* https://www.chromestatus.com/feature/5264933985976320

Although when reading:
https://www.w3.org/community/web-bluetooth/web-bluetooth-community-group-charter/
they don't seem to plan to cover e.g. RFCOMM channels initially.

Having a full web-bluetooth stack could help e.g. #180,

bjost2s, 2017-11-25:
Still only for chrome :-(
Any idea when or if other brothers will follow?

ensonic, 2017-12-07:
Here is an example how people use webbluetooth with the Espruino microcontroller. 
http://www.espruino.com/Web+IDE

rbudde, 2019-06-26:
not feasable at the moment

closed 2019-06-26 by rbudde