OpenRoberta/robertalab #288: use webusb to avoid the need for a helper program

ensonic, 2016-11-11:
webusb is a new api. This would allow us to talk to usb devices directly from the browser (js), without having an extra proxy.

Spec: https://wicg.github.io/webusb/

Here is the current status:
* https://www.chromestatus.com/feature/5651917954875392

Having a full web-usb stack could help e.g. #180,

tuxlego, 2016-11-23:
This feature is very welcome, because I'm still struggling connecting our NXT via Linux.

cvejoski, 2016-11-24:
@tuxlego did you build the USB program from the sources or did you install it as a package?

tuxlego, 2016-11-24:
I've installed it from the Chrome Extension menu, Extension: Open Roberta 1.4.0
https://chrome.google.com/webstore/detail/open-roberta/pgdcbhlffbmoegkehoookdmghakdjieh?utm_source=chrome-app-launcher-info-dialog

ensonic, 2017-01-27:
@tuxlego please create a new ticket for NXT and linux.

rbudde, 2019-06-26:
not feasable at the moment

closed 2019-06-26 by rbudde