OpenRoberta/robertalab #290: Ev3dev program cannot be restarted on a brick

filipwoz, 2016-11-11:
When I load a program to Ev3dev it runs immediately. After it is finished I try to run the Python file again by choosing it on the brick in the File Browser. But then some small characters appear for a moment and the program does not start

ensonic, 2016-11-12:
I noticed that the ev3dev system still has python python pointing to python3. I already pushed a workaround in 
https://github.com/OpenRoberta/robertalab-ev3dev/commit/de01b5060c7f81047d096739af806edc01c54b70

For now, the only thing you can do, is to change the first line to be

  #!/usr/bin/python3

I'll release an upgrade in a week.

filipwoz, 2016-11-14:
Where and how can I change the first line?
When you release the upgrade is it then directly available in the "https://lab.open-roberta.org/"?

ensonic, 2016-11-14:
@filipwoz you need to use a ssh client (like putty for windows) to login to the ev3. There all the programs you created are in the home directory (/home/robot). You use an editor (vi is pre-installed) to change them. This would really be a short gap measure.

For the update, when logged into the ev3, you run 

    sudo apt-get update; sudo apt-get upgrade

This can take quite some time though (depending on how recent the image is). There is another ticket to add UI for this and update things more selectively (and thereby quicker).

filipwoz, 2016-11-14:
Thanks for clarification. I would be very happy if this upgrade would be released before Friday ;-)

filipwoz, 2016-11-24:
Could you release this fix soon please?

ensonic, 2016-11-24:
Yes, plan is tomorrow. If you check git there are more fixes and I did a lot of testing.

ensonic, 2016-11-25:
@filipwoz https://github.com/OpenRoberta/robertalab-ev3dev/releases/tag/v1.6.0 will take a day or two before you can do apt-get update; apt-get upgade to get it

closed 2016-11-25 by ensonic