OpenRoberta/robertalab #291: Ev3dev via USB - needs a new token each time plugged in again

filipwoz, 2016-11-11:
Ev3dev connected via USB - when unplugged and plugged again needs a new token to be connected to open roberta again. It is quite annoying to enter it again and again in the browser.

ensonic, 2016-11-12:
Are you using usb networking? But, yes, reusing the token would be nice and the ev3dev connector supports it, but the openrobertalab does not. I'll discuss it with the co-developers.

bjost2s, 2016-11-15:
All other systems using the OpenRobertaUSB program, which is communicating to the lab server and guaranteeds a stable connection. You don't have to reconnect after unplugg and plug. We should check if we can extend this program

ensonic, 2017-01-23:
this has been fixed in OpenRoberta/robertalab-ev3dev#31

closed 2017-01-23 by ensonic