OpenRoberta/robertalab #292: Ev3dev global variable values set in functions are not available in other functions

filipwoz, 2016-11-11:
I have written a program which firstly evaluates the sensors and saves the results in variables and then executes a set of behaviors based on the variable values.
In order to make it more visible for the students I wanted to put these 2 stages of the program into 2 functions. Unfortunatelly, the values written to the variables by the first function seem not to be available in the second function.
Both program versions: with and without functions attached.

[Downloads.zip](https://github.com/OpenRoberta/robertalab/files/586694/Downloads.zip)

bjost2s, 2016-11-23:
Same issue in micro:bit (also python) 

Solution could be to add the global variable names to the functions in which they are used, so that we can declare them as global if needed. This could be done in blockly easily.

bjost2s, 2017-02-01:
@enconic, please comment on my proposal. I think we cannot fix it for 2.2. anymore, but we should discuss a solution, also for #292

ensonic, 2017-02-02:
Yes, each function will need to add a 'globals foo, bar' at the start. Ideally only for those vars it references.

closed 2017-03-28 by cvejoski