OpenRoberta/robertalab #293: OR Stand-alone-server problem with USB connection and EV3

Thorlogy, 2016-11-14:
We have recurring problems with the USB connection if we use the stand alone server. 
Within a 2 hour course the EV3 robot has to be reconnected 3 to 4 times. 

Reconnected means: 
- disconnect via USB program, 
- connect via USB program
- copy token
- and connect via OR Lab

bjost2s, 2016-11-15:
to which ip do you connect to your server in the usb program?
on which system does this appear?

dpyka, 2016-11-15:
By the way, the USB program stores a log file on your hard drive. I think on Windows it is somewhere in %APPDATA%. If you can provide the log file everytime you think the USB program did something wrong, we can investigate.

Thorlogy, 2016-11-21:
We used the chrome app and a chrome book.

bjost2s, 2016-11-21:
We have to check what are the differences to the usb program connection wich is working fine in this case

bjost2s, 2017-01-08:
What exactly was the error message? Can you provide a screenshot?

bjost2s, 2017-02-01:
@Thorlogy please reopen, if there is a way to reproduce this bug.

closed 2017-02-01 by bjost2s