OpenRoberta/robertalab #295: Lists: AddItem(itemvalue) / RemoveItemAt(position)

filipwoz, 2016-11-14:
It would be nice to have the following list functions available:
AddItem(itemvalue) adding a new item at the end of the list and returning the index of the new item
RemoveItemAt(position) removing an item from the given position of the list

bjost2s, 2016-11-15:
If you select the expert toolbox, there are some list blocks available. 
[image]

filipwoz, 2016-11-15:
ok the "get sub-list from" can serve as RemoveItemAt workaround.
But AddItem cannot be implemented with the existing functions

bjost2s, 2016-11-21:
[image]

Is this not working for you?

And what is the problem with this?
[image]

filipwoz, 2016-11-21:
OK I did not know that there are such additional functions in the combo box. 
But these are existing only in the EV3 version. I would need them ported also on NXT.

bjost2s, 2016-11-22:
Currently we do not support all of the list functions in NXT. The reason is that we have agreed to use the original lego firmware. With this firmware it is not possible to support more array functions like the EV3 does.

bjost2s, 2016-12-03:
Please reopen if this is still an issue

closed 2016-12-03 by bjost2s