OpenRoberta/robertalab #296: initial db setup

ensonic, 2016-11-14:
If someone forgets to copy the db or the db gets corrupted, one gets this exception:
https://gist.github.com/ensonic/c24c5e8a20a47f553aa8e8b1bc47110d

we should catch those and suggest to copy the db.

We also need to fix the ./ora.sh --reset-db command, this tries to to a git checkout, but it should copy the OpeRobertaServer/dbBase to OpenRobertaServer/db. 

Ideally ora.sh --start-server can do this if db is empty or not present?

ensonic, 2016-11-30:
Any objection, if I change --start-server to also copy the db if not present?

ensonic, 2016-12-02:
As discussed in the meeting, we should check if the database is good in the server and should terminate with a helpful error message. Then we don't need to modify ora.sh.

rbudde, 2016-12-09:
I started to change the ora.sh script. W.r.t. this issue I did the following:
- the **--reset-db** command copies the "default" database from dbBase (as suggested)
- the **--start-server** command is replaced by a **--start-from-git** command, which can only be called from the root of the working tree. With parameter **-r** it will reset the database to dbBase (I could make -r to the default, but I want to avoid data loss during long test sessions with many server starts).
- the **--export** command creates in the installation directory 2 sh scripts to start the server resp database server in two different processes (to achieve this, is the main reason of my changes to the ora.sh script).

Call **ora.sh --help** to see the commands available.

closed 2016-12-21 by bjost2s