OpenRoberta/robertalab #305: Code using functions does not compile for NXT

filipwoz, 2016-11-22:
When I try to transfer the attached program to NXT (simple sumo program with subfunctions
[FunctionsNXT.pdf](https://github.com/OpenRoberta/robertalab/files/606098/FunctionsNXT.pdf)
[SumoSimpleBlocks.zip](https://github.com/OpenRoberta/robertalab/files/606099/SumoSimpleBlocks.zip)

) I am getting an error that the program cannot be translated into the target machine code.
A similar program without functions (also attached) goes through.

bjost2s, 2016-11-23:
Seems that global variables are declared in main instead on top of the program, so that they are not available in the functions

``
task main() {
  SetSensor(S1, SENSOR_TOUCH);
  SetSensor(S2, SENSOR_SOUND);
  SetSensor(S3, SENSOR_LIGHT);
  SetSensor(S4, SENSOR_LOWSPEED);
  bool IsPressed = false;
  bool IsLine = false;
  while ( true ) {
    actions();;
    sensors();;
  }
}
``

filipwoz, 2016-11-23:
Yes, I think it was the problem: I could even transport the program to the brick but by setting a variable in one function I could not see that the other function reacts on this value.

elhele, 2016-11-23:
That might be a good idea to pass the variables we need to change to the functions with a pointer, not just to change the globals. In this case we won't have to move the variables from the main. Or would it be too hard to change the block and we don't want it?

bjost2s, 2016-11-23:
I think this bug is only about global variables. We should put the declaration just after the define statements:
```
#define WHEELDIAMETER 5.6
#define TRACKWIDTH 12.0
#define MAXLINES 8 
#include "NEPODefs.h" // contains NEPO declarations for the NXC NXT API resources

bool IsPressed = false;
bool IsLine = false;
 
void sensors() {
  if ( Sensor(S2) || Sensor(S3) ) {
    IsPressed = true;
  }
  if ( SensorColor(S1, "COLOR") == INPUT_BLACKCOLOR ) {
    IsLine = true;
  } else if ( true ) {
    IsLine = false;
  }
}
 
void actions() {
  if ( IsLine ) {
    RotateMotorEx(OUT_BC, -1 * 50, (10 * 360 / (PI * WHEELDIAMETER)), 0, true, true);
    Wait(1);
    RotateMotorEx(OUT_BC, 50, (120 * TRACKWIDTH / WHEELDIAMETER), 100, true, true);
    Wait(1);
    IsPressed = false;
  } else if ( IsPressed ) {
    OnFwdReg(OUT_BC, 100, OUT_REGMODE_SYNC);
  } else {
    OnFwdReg(OUT_BC, 50, OUT_REGMODE_SYNC);
  }
}
 
task main() {
  SetSensor(S1, SENSOR_TOUCH);
  SetSensor(S2, SENSOR_SOUND);
  SetSensor(S3, SENSOR_LIGHT);
  SetSensor(S4, SENSOR_LOWSPEED);
  while ( true ) {
    actions();
    sensors();
  }
}
```

VinArt, 2016-11-24:
@bjost2s if you move this declaration to being global how do I define a variable that would be in main then? Also it is programming 101: passing parameters to functions. So the correct way would be:
`bool isPressed = false;
  bool isLine = false;
  while ( true ) {
    actions();;
    sensors(isPressed, isLine);;
  }`
The only problem remaining is to set a flag in blocks like mutable or not, so the parameters will be passed by reference and not just copied. Or write the code so it would return muted vars.

BTW, if you include globals, don't forget to add GOTO, that would spice up the code =)))

bjost2s, 2016-11-24:
@VinArt in general you are right, but we do not generate real c code but nxc, not exactly c. Maybe this [http://bricxcc.sourceforge.net/nbc/nxcdoc/nxcapi/vars.html](http://bricxcc.sourceforge.net/nbc/nxcdoc/nxcapi/vars.html) helps to understand it.

I attached a simple test program for what @elhele fixed, it works.
[NEPOprog.xml.zip](https://github.com/OpenRoberta/robertalab/files/611949/NEPOprog.xml.zip)

btw there is another bug with the procedure blocks, see #312

VinArt, 2016-11-24:
@bjost2s I do understand the difference between a local and a global variable, I even see no problem using them here, since it is a playground. What I see as a problem is who will be using this playground, right, kids, and teachers. And I think it is a not-so-great-for-an-idea to teach kids to use global variables starting from school, they will have their chances later on. There exist languages that do not even support globals. I think that avoiding globals will also relieve you from a pain synchronising the state of those.

maybe this http://bricxcc.sourceforge.net/nbc/nxcdoc/nxcapi/func.html helps to understand it.

bjost2s, 2016-11-26:
Please use out google group for general ideas and features.https://groups.google.com/forum/#!forum/open-roberta
bug is fixed

closed 2016-11-26 by bjost2s