OpenRoberta/robertalab #307: As a teacher I would like to setup a local Open Roberta Lab server on a raspberry pi and maintain it easily, eg. with a GUI

bjost2s, 2016-11-22:
make it as easy as possible:
- [ ] install the server on a raspi pi and / or provide an image with the already installed server
- [ ] provide start, stop, restart server commands
- [ ] update the server, if desired. This includes also 
  * update the server to a specific version
  * update the database 
  * update robots firmware, if necessary and possible
- [ ] offer a basic fail-safe system
- [ ] offer a basic server configuration setup
  * add / remove robot plugins
  * configure toolboxes for robot plugins? 

nice to have:
- [ ] synchronise the raspi pi data base with the online data base on demand

bjost2s, 2016-11-22:
Please discuss ideas for this tool / script / system, to make it as usefull as possible.

ensonic, 2016-11-23:
I have a startup script already. I can submit t, but we need to decide where it should go. What about: https://github.com/OpenRoberta/robertalab/tree/master/OpenRobertaServer inside a systemd folder?

bjost2s, 2016-11-23:
I think we should have a separate project for this. If the user can already use git and find the script it is mayby useless. 
In best case the user downloads something or it is already on the raspi image, that he can execute and then starts with downloading (via git) the lab.

We already have the ora script to start the server, handle database and export stuff, so maybe it could be part of the new program/system.

@ensonic, could you attach your script to this ticket?

ensonic, 2016-11-24:
I've uploaded it here:
https://gist.github.com/ensonic/749d259f48302a2ed5b480c6e3a28930

ensonic, 2016-12-02:
I've adding it to OpenRobertaServer, so that it is in git from which we are going to build the debian package. The systemd init script is using the ora.sh script.

bjost2s, 2017-02-01:
@rbudde please have a look, if we could mark something as done

VinArt, 2017-11-27:
We could probably further automate procedures like update server with systemd with one-shot services and then use Cockpit web ui for managing them.