OpenRoberta/robertalab #322: Can't create a new variable from the variables menu

drigz, 2016-12-02:
You have to create a variable by clicking the plus on the "start" block. This isn't clear to all users, some of whom try to create one from the Variables menu.

If you open the Variables menu on an empty program on OpenRoberta, you see nothing (which looks a bit broken):

[image]

Expected behaviour: a "Create variable" button as for [upstream Blockly](https://blockly-demo.appspot.com/static/demos/code/index.html):

[image]

bjost2s, 2016-12-03:
if we would do that, the create variable block could be placed somewhere in the editor. What if you declare a variable and use it before. The program will no longer be compilable or crash at runtime. The plus is a NEPO feature, explained in our block description.

drigz, 2016-12-05:
That's not a block, it's a button. You can't drag it into your program, you just click it and it asks for the name of the new variable. You can try it on the [Blockly web playground](https://blockly-demo.appspot.com/static/demos/code/index.html).

bjost2s, 2017-01-08:
ok, this is new in blockly. But I think it is not straight forward having buttons in the flyout. As far as I now, it is the only case where a button appears in the flyout. 
We would need a new input modal because we need at least the variable name and the type. 
How should this look for loop or procedure variables? In the popup we have to make the connection to the specific global, local or loop variable. I think this is much more difficult to understand than the way we have it currently. 
In former times I opened the help files when the user clicked on the (empty) variable toolbox category. Wouldn't it be more helpful?

bidaian, 2017-02-21:
Thanks @drigz, I found out how to create variables thanks to this issue. I like the new solution in blockly.

@bjost2s from your comment I don't understand the the current relationship between blockly and openroberta. Do you still pull from blockly or is it like two separate projects now?

closed 2017-11-25 by bjost2s