OpenRoberta/robertalab #328: Break or continue within wait-for block leads to compiler error (at least) in EV3

maweki, 2016-12-02:
[image]

This creates the code

```
    public void run() throws Exception {
 
        if ( true ) {
            while ( true ) {
                while ( true ) {
                    if ( false ) {
                        break;
                        break;
                    }
                    if ( false ) {
                        continue;
                        break;
                    }
                    hal.waitFor(15);
                }
            }
        }
    }
```

The `break`s behind the `break`/`continue` are unreachable code and lead to a compilation error for java. Other languages might not be a problem from a compilation point of view (js/python would just ignore unreachable code).

But semantically this should break out of the outer loop and not the inner (that the user is not aware of). This might be (is probably) a seperate issue.

ensonic, 2016-12-08:
Also applies to python

    def run():
        while True:
            while True:
                if False:
                    break
                    break
                if False:
                    continue
                    break
                hal.waitFor(15)

In addition, the hal.waitFor(15) is unexpected (it is generated as part of the waitFor block).

maweki, 2016-12-08:
@ensonic the semantic error (not breaking out of the expected loop) applies to python. Unreachable code should not be a problem, right?

ensonic, 2016-12-08:
Yes, I just wanted to confirm that we have the same issues with the generated python code structure wise.

maweki, 2017-01-03:
The compiler error can easily be solved by replacing every `break` with `if (true) break;` and every `continue` with `if (true) continue;`. The code after that will then be _dead code_ instead of _unreachable code_ and is only a warning instead of a compiler error.

The semantic error remains.

maweki, 2017-01-03:
Also, since java 7, loops can be labeled and they can be broken out of with that label.

* http://stackoverflow.com/questions/3821827/loop-in-java-code-what-is-this-why-does-it-compile
* http://docs.oracle.com/javase/specs/jls/se7/html/jls-14.html#jls-14.7

The only problem is, that labels can't be duplicated within the same nested loop block. So for every loop block there needs to be a unique label and every `break` would break out of the innermost loop-block.

This isn't necessary for the wait-for-blocks, I think, since their `break`/`continue` is always the correct corresponding one.

closed 2017-04-18 by cvejoski