OpenRoberta/robertalab #329: Variable names can be reserved words

maweki, 2016-12-06:
I can set variable names to reserved words in the OpenRobertaLab which leads to uncompilable code and no error message. I think there should be an exclamation-mark by the chosen variable name with an explanation.

[image]

This should be a problem for all supported languages.

bjost2s, 2016-12-06:
Unfortunately we allow in the blocks to use reserved words, but before the code compilation (not executed for show code) we check for reserved words and the user get a (not really helpful) message.
If we don't allow reserved words in blockly we should take into account, that the check will go over all programming languages used (currently java, python, c++, c) in the Open Roberta Lab, eg for EV3 also "def" won't be allowed.

maweki, 2016-12-07:
I think it would be fine to have the union of the reserved words of all supported languages as not-allowed identifiers. Is this information we should hide (from a learning-perspective), that languages *do* indeed have reserved words that I am not allowed to use as identifiers?

bjost2s, 2017-01-08:
We have now language list in blockly, where the variable and procedure names can be validated against.
TODO extend these list with names generated from us, that are not allowed.

maweki, 2017-01-09:
The Python reserved word list is not correct. This is the python 2.7 list. MicroPython, which is running on the bbc micro bit is based on Python 3 which has different keywords.

https://docs.python.org/3.4/reference/lexical_analysis.html#keywords

 `False`, `None`, and `True` got promoted to keywords while `print` went from keyword to function.

ensonic, 2017-01-10:
@maweki thanks for checking, could you send a PR with the fixes?

maweki, 2017-01-10:
@ensonic sure. Just to make sure: Is the MicroPython target static or are there any other robots/implementations/interfaces in the works that might target 2.7 so that we'd want the union of both sets of keywords?

ensonic, 2017-01-10:
@maweki: lego ev3 via ev3dev.org uses python3 as well

maweki, 2017-01-12:
@ensonic I was just working on that and I saw that apparently the python that targets NAO is 2.6/2.7

http://doc.aldebaran.com/1-14/dev/python/install_guide.html#python-install-guide

Since python2 and python3 are in some ways different languages, would we have two sets of keywords for py2 and py3 respectively or have a union of both?

ensonic, 2017-01-12:
Lets make it a union to play safe. I'll check if we can move nao to python3 as well.

maweki, 2017-01-12:
Do we want to use this mechanism as well to solve #333 by pretending that the imports we use are keywords?

JanisM91, 2017-01-12:
@ensonic The SDK for NAO only supports Python 2.6/2.7 and the python runtime on the robot is also 2.7. It is part of the NAOqi so we can not change the version to python3.

maweki, 2017-01-12:
I added the keywords and split the imports and variable names that (i think) belong to ev3dev and added two more exceptions.

closed 2017-01-27 by bjost2s