OpenRoberta/robertalab #331: can we change visitMainTask() to be more like visitMethodVoid()

ensonic, 2016-12-06:
for #319 I need to check if the body of the main function is empty. Right now this is not doable in visitMainTask() since this is not traversing into the body. In visitMethod*() we call getBody().visit(this); where getBody() returns a StmtList.

bjost2s, 2016-12-06:
if think the easiest way to do this is after completing #308, because than you can do it in the ev3dev plugin which is not affecting all the other plugin implementations.

ensonic, 2016-12-06:
We probably need the same feature for the microbit (python).

cvejoski, 2016-12-07:
In this case we should change the MainTask block to be like function and then we can have the possibility of getBody().

One simple solution (I do not if it is the right one) is using a different visitor to check if the MainTask is empty

cvejoski, 2016-12-07:
For the microbit (python) the program is just a script with no main method and no run()

ensonic, 2016-12-07:
If it is too complicated to add a getBody() into the visitMainTask(), we could also add a boolean isProgramEmpty() method. Code generators like java can ignore this.

ensonic, 2016-12-08:
Works great. Thanks!

closed 2016-12-08 by ensonic