OpenRoberta/robertalab #334: Variable definitions can be the names of existing variables

maweki, 2016-12-07:
This is similar to #333 but is specific to Java. I can define variables for my program that have names of already used variables in the class of the program, which makes the program uncompilable.

[image]

`main` and `run` are not a problem, since functions aren't objects, but I am not allowed to name a variable `hal`. This can either be solved by not allowing those names or maybe hiding this stuff (it's private anyway) in an abstract base class. This would lead (thanks Java) to multiple files for a program.

bjost2s, 2016-12-21:
we should check the user defined names before generation own once.

bjost2s, 2017-02-01:
I think we fixed it with #333?
If so, please close this issue.

maweki, 2017-02-01:
We kinda did for python. We have to recheck all imports across the board whether we're hiding anything.

closed 2017-02-11 by bjost2s