OpenRoberta/robertalab #336: Junit problems with class BlockTypeContainer

rbudde, 2016-12-09:
class BlockTypeContainer has a static map, that is implicitly reused between tests and thus generates trouble.

- Execute all tests from project RobotEv3. This succeeds, because from the RobotFactory-init blocks are added to the static maps (and the map is reused by later tests).
- Execute the single test AssignmentStmtTest. This fails, because the "VAR" block is not in the static map.

maweki, 2017-02-03:
The static keyword in tests might be critical in general. Setup-stuff should be final/immutable or just re-set-up it every time or is it time-critical?

ensonic, 2017-02-03:
For reference:
https://github.com/OpenRoberta/robertalab/blob/HEAD/OpenRobertaRobot/src/main/java/de/fhg/iais/roberta/syntax/BlockTypeContainer.java#L19

rbudde, 2018-03-09:
Commit bf497ddf8f929d6cbb45e931192448639c52ed51 solves the problem in the unit tests. This bug was reported as #774 too. The static variable (in the productive code, not in the test!) should be removed and an object injected

closed 2018-04-04 by bjost2s