OpenRoberta/robertalab #338: robot selection dialog is broken

ensonic, 2016-12-16:
git is at:
commit 4d1318fb9089206d2d4df2190fa191b7ee04b06f
Author: Reinhard Budde <reinhard.budde@iais.fraunhofer.de>
Date:   Fri Dec 16 10:49:33 2016 +0100

    #337 send robotd as array instead of map to browser. Added "sim" explicitly

[image]

In the log I see
```
2016-12-16 15:57:33,804 [qtp1486491935-10] INFO d.f.i.r.j.provider.OraDataProvider - session #1 created
2016-12-16 15:57:33,810 [qtp1486491935-10] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 7 msec: page init
2016-12-16 15:57:33,810 [qtp1486491935-10] INFO d.fhg.iais.roberta.util.ClientLogger - 1 log entry written
2016-12-16 15:57:33,810 [qtp1486491935-10] INFO d.f.i.r.j.r.all.ClientAdmin - command is: init
2016-12-16 15:57:33,812 [qtp1486491935-10] INFO d.f.i.r.j.r.all.ClientAdmin - success: create init object
2016-12-16 15:57:33,877 [qtp1486491935-12] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 0 msec: init gui state model
2016-12-16 15:57:33,877 [qtp1486491935-12] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init gui state
2016-12-16 15:57:33,877 [qtp1486491935-12] INFO d.fhg.iais.roberta.util.ClientLogger - 2 log entries written
2016-12-16 15:57:33,877 [qtp1486491935-12] INFO d.f.i.r.j.r.all.ClientAdmin - command is: setRobot
2016-12-16 15:57:33,877 [qtp1486491935-12] INFO d.f.i.r.r.RobotCommunicator - token 1Q2W3E4R is not waited for. Ok.
2016-12-16 15:57:33,878 [qtp1486491935-12] INFO d.f.i.r.j.r.all.ClientAdmin - set robot to ev3
2016-12-16 15:57:33,891 [qtp1486491935-16] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[ERR ]] [[TIME]] 1 msec: set robot 'ev3', then EXCEPTION: TypeError: Cannot read property 'disable' of undefined
2016-12-16 15:57:33,891 [qtp1486491935-16] INFO d.fhg.iais.roberta.util.ClientLogger - 1 log entry written
2016-12-16 15:57:33,907 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init robot forms
2016-12-16 15:57:33,908 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - 1 log entry written
2016-12-16 15:57:33,908 [qtp1486491935-32] INFO d.f.i.r.j.r.all.ClientUser - command is: clear
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init user forms
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init program list view
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init program delete
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init configuration view
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init configuration delete
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init program sharing view
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init log list view
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init configuration view
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init program view
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] init menu view
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] result from server: {"cmd":"clear","rc":"ok","serverTime":"Fri Dec 16 15:57:33 CET 2016","server.version":"2.1.1"}
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 5 msec: clear user
2016-12-16 15:57:38,005 [qtp1486491935-32] INFO d.fhg.iais.roberta.util.ClientLogger - 12 log entries written
```

ensonic, 2016-12-16:
It was still good at 8c8eac984fa0b114950a5e5719322e1d0464fe99

bjost2s, 2016-12-16:
fixed in #337

closed 2016-12-16 by bjost2s