OpenRoberta/robertalab #340: Problems with a link for a driver in "EV3 (leJOS) driver installation under Windows 8, 10 instructions"

elhele, 2016-12-20:
The first step in the "EV3 (leJOS) driver installation under Windows 8, 10" instruction (https://mp-devel.iais.fraunhofer.de/wiki/display/ORInfo/EV3+%28leJOS%29+driver+installation+under+Windows+8%2C+10)  is "First of all, download the following driver from the official Microsoft page (popups must be allowed, otherwise the download-dialog does not appear):
http://catalog.update.microsoft.com/v7/site/ScopedViewRedirect.aspx?updateid=37e35bd4-d788-4b83-9416-f78e439f90a2"
However, it is not possible to download the drivers using this link:
[image]

axel-dumschat, 2016-12-20:
Hallo  elhele,

die Gültigkeitsdauer der Links ist scheinbar befristet. 

Ruf doch mal die folgende Suchseite auf: http://www.catalog.update.microsoft.com/home.aspx
und gib den Suchbegriff "USB Ethernet/RNDIS Gadget" ein.

Der Treiber 5 bzw. 6 müsste es sein.

Ich habe den Treiber 5 in den Anhang gepackt.
[rndis.zip](https://github.com/OpenRoberta/robertalab/files/664972/rndis.zip)

Viel Erfolg,

Axel

bjost2s, 2016-12-21:
Please send issues for the open roberta wiki to roberta-zentrale@iais.fraunhofer.de

Thank you for your understanding. We only track lab issues here.

closed 2016-12-21 by bjost2s

elhele, 2017-01-05:
Hello Beate,

I remember, you mentioned that we need to fix a pyhton program that  
was written by Max. Could you send it to me so I could look into it?

Regards,
Evgeniya