OpenRoberta/robertalab #344: Ardu project cleanup

bjost2s, 2016-12-21:
While copiing the project for BOB3 I found files / folders named EV3... Are they used?
Please delete or rename them

elhele, 2016-12-27:
The files seem to be automatically generated by Ev3Configuration.g4 configuration files. They exist not only in arduino, but in all other robots´ projects.

ensonic, 2016-12-27:
Hmm, should we even have those files outside of ./RobotEV3
```
> find . -name "Ev3Configuration.g4"
./RobotMbed/src/main/antlr4/de/fhg/iais/roberta/ev3Configuration/generated/Ev3Configuration.g4
./RobotEV3/src/main/antlr4/de/fhg/iais/roberta/ev3Configuration/generated/Ev3Configuration.g4
./RobotNAO/src/main/antlr4/de/fhg/iais/roberta/ev3Configuration/generated/Ev3Configuration.g4
./OpenRobertaRobot/src/main/antlr4/de/fhg/iais/roberta/ev3Configuration/generated/Ev3Configuration.g4
./RobotNXT/src/main/antlr4/de/fhg/iais/roberta/ev3Configuration/generated/Ev3Configuration.g4
./OpenRobertaServer/src/main/antlr4/de/fhg/iais/roberta/ev3Configuration/generated/Ev3Configuration.g4
./RobotArdu/src/main/antlr4/de/fhg/iais/roberta/ev3Configuration/generated/Ev3Configuration.g4
```

elhele, 2016-12-29:
@ensonic  they generate some files those are used in ArduConfigurationParseTree2ArduConfigurationVisitor and in NxtConfigurationParseTree2NxtConfigurationVisitor. In other projects they are not used. However, those visitor files also don't seems to be used anywhere else and their removal doesn't affect anything. They must have been just copied from EV3 and renamed. So, I´ll remove them.

closed 2017-01-06 by elhele