OpenRoberta/robertalab #345: BOB3 Integration

bjost2s, 2016-12-21:
BOB3 is a small programmable board, like mirco:bit or calliope, based on an arduino
http://www.bob3.org/

 - [ ] BOB3  code generation
 - [ ] BOB3 compiler workflow
 - [ ] BOB3 GUI blocks and toolbox
 - [ ] BOB3 Adopt simulation 
 - [ ] BOB3 adopt USB program 

arduino libraries are available

VinArt, 2017-08-17:
An icon is needed for Bob3 menu items and start screen.

VinArt, 2017-09-04:
@bjost2s 52b9b00d010d2bb41a5b5511771bb000657667e6 gives an exception de.fhg.iais.roberta.util.dbc.DbcException: There was no correct configuration block found! robControls_loopForever_ardu

If I change it to the robBrick_EV3-Brick that I understand is wrong, there are no exceptions...

bjost2s, 2017-09-25:
Please do not close this ticket before it is labeled as resolved

bjost2s, 2017-09-25:
Please check, if it is possible to instantiated the timer only when the timer is used in the wait-for block

bjost2s, 2017-09-25:
GUI is logging every second (or two) list success in comm.js function listRobotsFromAgent
although I switched to EV3lejos!
[image]

closed 2017-10-02 by cvejoski