OpenRoberta/robertalab #347: NAO brick configuration

JanisM91, 2016-12-21:
Check what configuration values are necessary to make sure all used functions from NAO can be executed. The NAO is not modular. Therefore neither actors nor sensors need to be added to the configuration.

Provide a new configuration in the database with

-   new brick name (NAO)
- no actors
- no sensors
- fields for ip address, password and username

bjost2s, 2017-01-24:
The configuration in general is there to make sure the program "knows" what sensors and actors are plugged in at which port. For NAO we cannot exchange sensors or actors, or?

JanisM91, 2017-01-24:
It is not possible to exchange sensors or actors for NAO.
Do you recommend to use it to inform about the sensors and actors with a picture or would you prefer to make a Block and add undeletable Blocks that represent the sensors and actors?
I attached two examples out of the NAO documentation to this comment.

[image]

[image]

closed 2017-02-04 by bjost2s