OpenRoberta/robertalab #356: EV3 Simulator - sensor data not shown

mkany, 2017-01-10:
hello, 
if I change the scene in firefox or chrome  browser the sensor data are not shown. 
Then I press Strg + ans Strg 0, all sensor data are shown.

greetings
Michael Kany

bjost2s, 2017-01-24:
Thanks, we did some restructuring on the gui side and it seems that this bug is fixed with it.

mkany, 2017-01-24:
Hello, 

Ist this fix released?

it seems that this bug isn’t fixed now.

I tested it with Win10, Firefox and Chrom Browsers.

Maybe my resolution is 1920x1080?

 

Greeetings

Michael Kany

 

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

bjost2s, 2017-01-27:
It is not released yet. The next release is scheduled for the 8th of february.

closed 2017-02-25 by bjost2s