OpenRoberta/robertalab #360: Error in test while running mvn install

bidaian, 2017-01-16:
After cloning, if I try to run mvn clean install I get an error in a test. Does anybody know what could be wrong? (Debian Jessie, mvn 3.3.9)

Tests run: 53, Failures: 0, Errors: 53, Skipped: 0, Time elapsed: 0.098 sec <<< FAILURE! - in de.f
hg.iais.roberta.factory.generic.RobotModeFactoryTest
getListElementOperationFromString(de.fhg.iais.roberta.factory.generic.RobotModeFactoryTest)  Time 
elapsed: 0.008 sec  <<< ERROR!
de.fhg.iais.roberta.util.dbc.DbcException: Assertion notNull is violated
	at de.fhg.iais.roberta.util.dbc.Assert.notNull(Assert.java:179)
	at de.fhg.iais.roberta.util.RobertaProperties.getRobotNumberFromProperty(RobertaProperties
.java:87)
	at de.fhg.iais.roberta.factory.EV3Factory.<init>(EV3Factory.java:51)
	at de.fhg.iais.roberta.factory.generic.RobotModeFactoryTest.<init>(RobotModeFactoryTest.ja
va:30)

invalidListElementOperation(de.fhg.iais.roberta.factory.generic.RobotModeFactoryTest)  Time elapse
d: 0.001 sec  <<< ERROR!
de.fhg.iais.roberta.util.dbc.DbcException: Assertion notNull is violated
	at de.fhg.iais.roberta.util.dbc.Assert.notNull(Assert.java:179)
	at de.fhg.iais.roberta.util.RobertaProperties.getRobotNumberFromProperty(RobertaProperties
.java:87)
	at de.fhg.iais.roberta.factory.EV3Factory.<init>(EV3Factory.java:51)
	at de.fhg.iais.roberta.factory.generic.RobotModeFactoryTest.<init>(RobotModeFactoryTest.ja
va:30)

cvejoski, 2017-01-17:
the build is for the master or develop branch?

bidaian, 2017-01-18:
It is for master. I tried on another jessie box and it works ... I guess it's a library problem. Any idea of which library/program could be causing that?

cvejoski, 2017-01-19:
I am not sure but it is possible to be JUnit. I have also tested in on Debian 8 with java 1.8 and maven 3.3.9 and there was no problem, but on Fedora 25 and Arch the same problem. 

Some of the property files are not loaded correctly during the build and that is why the tests are failing.

cvejoski, 2017-01-20:
fixed with: https://github.com/OpenRoberta/robertalab/commit/dd68a3cf11f2e54582593be087e656d87b983ea1

closed 2017-01-20 by cvejoski

bidaian, 2017-01-20:
It looks better, I still get this (develop). Do you need more context?

11:45:17.493 [main] INFO  d.f.i.r.testutil.JavaBeanTester -   property : i
11:45:17.494 [main] ERROR d.f.i.r.testutil.JavaBeanTester - Expected value [1] and actual value [2] differ for de.fhg.iais.roberta.testutil.test.BeanTesterFail1#i
11:45:17.494 [main] INFO  d.f.i.r.testutil.JavaBeanTester -   property : s
11:45:17.494 [main] INFO  d.f.i.r.testutil.JavaBeanTester -   property : t
11:45:17.495 [main] INFO  d.f.i.r.testutil.JavaBeanTester - CLASS UNDER TEST (SUT) de.fhg.iais.roberta.testutil.test.BeanTesterFail2
11:45:17.498 [main] INFO  d.f.i.r.testutil.JavaBeanTester -   property : i
11:45:17.498 [main] INFO  d.f.i.r.testutil.JavaBeanTester -   property : s
11:45:17.498 [main] INFO  d.f.i.r.testutil.JavaBeanTester -   property : t
11:45:17.499 [main] INFO  d.f.i.r.testutil.JavaBeanTester -   immutable check for: setS
11:45:17.499 [main] ERROR d.f.i.r.testutil.JavaBeanTester - a setter of an immmutable object succeeded for the setter de.fhg.iais.roberta.testutil.test.BeanTesterFail2#setS
11:45:17.499 [main] INFO  d.f.i.r.testutil.JavaBeanTester -   immutable check for: setT
11:45:17.499 [main] ERROR d.f.i.r.testutil.JavaBeanTester - a setter of an immmutable object succeeded for the setter de.fhg.iais.roberta.testutil.test.BeanTesterFail2#setT
11:45:17.500 [main] INFO  d.f.i.r.testutil.JavaBeanTester -   immutable check for: setI
11:45:17.500 [main] ERROR d.f.i.r.testutil.JavaBeanTester - a setter of an immmutable object succeeded for the setter de.fhg.iais.roberta.testutil.test.BeanTesterFail2#setI