OpenRoberta/robertalab #364: Downloaded source code contains HTML entities

hlg, 2017-01-19:
When downloading source code it contains HTML entities, e.g. in Java:

    private Set&lt;UsedSensor&gt; usedSensors = new LinkedHashSet&lt;UsedSensor&gt;();

instead of:

   private Set<UsedSensor> usedSensors = new LinkedHashSet<UsedSensor>();

This would be correct for code display in browser, but not for downloading a source code file.

cvejoski, 2017-01-20:
The file extension is not correct of the downloaded file. For example,  the programing language for Calliope is C++ but the downloaded source file is with **java** extension.

ensonic, 2017-01-21:
Just got another comment on irc, that in python the quotes are gone:
```
When i download the python file i get:
if __name__ == &quot;__main__&quot;:
However, the same python file in D:\temp\openroberta\userProyects is correct:
if __name__ == "__main__":
```

closed 2017-01-24 by cvejoski