OpenRoberta/robertalab #367: Troubles with local openrobertalab server: "D:\temp\openroberta\userProyects"

DJuego, 2017-01-21:
Hi! Here a trouble!

My settings is: 

>Windows 10 x64
> 10 laptops
> 10 EV3 bricks with ev3dev OS

I want to launch in each laptop a local openrobertalab server. I have used succesfully the procedure described in the README.md. I make a "./ora.sh --start-from-git "

The connection will be USB (not WIFI). I can connect with success. The conection seems very stable. 
However i can not run the programs in EV3."

I receive in openrobertalab the message: _"The program could not be saved because of an internal error."_
The cause has been identified: 

openrobertalab try to create the path "D:\temp\openRoberta\userProjects\FEK477QX\src'" but the laptops have not a drive with a letter unit D:

The openrobertalab log is here (http://pastecode.org/index.php/view/64658768)

 _"D:\temp\openRoberta\userProjects\FEK477QX\src' could not be created"_

Why "D:" and not "C:"? Is it possible to change the path? How? I can modify the source if it is necesary?

DJuego

PS: I am using jdk-8u112 "portable" (no installation)

ensonic, 2017-01-21:
The tempdir code seems to be from https://github.com/OpenRoberta/robertalab/blob/b339e42fdaa86183e3aa715612054114bb0c8275/OpenRobertaRobot/src/main/java/de/fhg/iais/roberta/util/RobertaProperties.java#L58-L67

cvejoski, 2017-01-21:
You can change the location of the temporary folder in the property file:

https://github.com/OpenRoberta/robertalab/blob/master/OpenRobertaServer/src/main/resources/openRoberta.properties#L12

Don't forget to build again the project

ensonic, 2017-01-21:
It is a regression from
https://github.com/OpenRoberta/robertalab/commit/30dcb5448bd5a35d95dc035628011630e3351413#diff-5e6a54a0ee6f57428ce011298ea6113f

We should not have a default in the property file, because then the code will never consider the actual java tempdir.

ensonic, 2017-01-21:
Already fixed in develop:
https://github.com/OpenRoberta/robertalab/commit/0a8a6e0464c9ce5af25ebe2f8d75ae7d535e0564

closed 2017-01-21 by ensonic