OpenRoberta/robertalab #370: Block show picture in EV3 is neither evaluating x nor y values

bjost2s, 2017-01-24:
[image]

We should discuss what to do
-  remove input fieds x and y
-  implement x and y (shifting the image)

bjost2s, 2017-02-01:
we decided to remove the input fields.
If we once have new block for custom pictures it probably will make sence to offer them there, because it will then be possible to show smaller pictures.

bjost2s, 2017-02-04:
Removing seems to be difficult, because old programs causes problems: x and y math blocks cannot connect to show block.

bjost2s, 2017-03-25:
ORA-2.2.1 will not have an update for the EV3, so moved to ORA-2.3.0

bjost2s, 2017-09-12:
We will create a new block without x and y that will substitude the old one. For backwards compatibility the old one will still work and create code, but it won't be available in the toolbox anymore.

bjost2s, 2017-09-12:
fixed here b5c59970d02c69353fa9cfb337cba86c648a8da6 applied wrong ticket number #307

ensonic, 2017-09-15:
Tested on the ev3dev:
- import old program into the git HEAD version: okay
- use the new block without x/y in the HEAD version: okay

closed 2017-09-15 by ensonic