OpenRoberta/robertalab #389: Connect NXT to Linux

tuxlego, 2017-02-05:
I don't own a Windows system and so I want to connect my NXT to a Linux system (using Ubuntu 16.04 LT and RaspberryPI incl. OpenRobertaServer).

- I've tried the Chrome Extension and  installed it from the Chrome Extension menu, Extension: Open Roberta 1.4.0; [https://chrome.google.com/webstore/detail/open-roberta/pgdcbhlffbmoegkehoookdmghakdjieh?utm_source=chrome-app-launcher-info-dialog --> but doesn't work](url)

- I've tried your new OpenRobertaUSB for Linux, but get the following error:
`Feb 05, 2017 9:57:15 PM lejos.pc.comm.NXTConnector search
SCHWERWIEGEND: Error: Failed to load USB comms driver. lejos.pc.comm.NXTCommException: lejos.pc.comm.NXTCommException: Cannot load NXTComm driver`

cvejoski, 2017-02-06:
For using the NXT and OpenRobertaUSB under linux you have to use java 32 bit version and also you have to install the following 32 bit libraries:

sudo apt-get install libc6-i386 libxext6:i386 libxrender1:i386 libXtst6:i386 libXi6:i386 libusb-0.1-4:i386

Also until the end of the week we will have installer OpenRobertaUSB for Linux

cvejoski, 2017-02-07:
Please use this USB Program: https://mp-devel.iais.fraunhofer.de/wiki/download/attachments/9503677/OpenRobertaUSBv2_0_4.tar.gz?version=1&modificationDate=1486387437369&api=v2

tuxlego, 2017-02-07:
thanks for the installer, but still not able to connect.

_steffen@lenovix:/usr/share/OpenRobertaUSB/java/bin$ java -jar -Dfile.encoding=utf-8 ../../OpenRobertaUSB.jar
07/02/2017 09:05:34.244 - [de.fhg.iais.roberta.usb.Main.configureLogger] - [INFO] - Logging to file: /home/steffen/OpenRobertaUSB/OpenRobertaUSB.log
07/02/2017 09:05:34.422 - [de.fhg.iais.roberta.usb.Main$1.getLocals] - [CONFIG] - Language de
07/02/2017 09:05:34.425 - [de.fhg.iais.roberta.connection.EV3USBConnector.<init>] - [CONFIG] - Robot ip 10.0.1.1
OpenJDK 64-Bit Server VM warning: You have loaded library /tmp/libjlibnxt7414171082747786742.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
Feb 07, 2017 9:05:35 PM lejos.pc.comm.NXTConnector search
SCHWERWIEGEND: Error: Failed to load USB comms driver. lejos.pc.comm.NXTCommException: lejos.pc.comm.NXTCommException: Cannot load NXTComm driver
07/02/2017 09:05:38.280 - [de.fhg.iais.roberta.connection.EV3USBConnector.findRobot] - [INFO] - No EV3 device connected yet_

_steffen@lenovix:~$ usb-devices
T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  6 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=0694 ProdID=0002 Rev=00.00
S:  SerialNumber=0016530C6DBC
C:  #Ifs= 1 Cfg#= 1 Atr=c0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)_

Rechecked installation:
_steffen@lenovix:/usr/share/OpenRobertaUSB$ sudo sh install.sh 
Holen:1 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]
OK:2 http://de.archive.ubuntu.com/ubuntu xenial InRelease 
OK:3 http://de.archive.ubuntu.com/ubuntu xenial-updates InRelease              
OK:4 http://de.archive.ubuntu.com/ubuntu xenial-backports InRelease            
Es wurden 102 kB in 0 s geholt (232 kB/s).                   
Paketlisten werden gelesen... Fertig
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
»libusb-0.1-4:i386« ist bereits die neuste Version (2:0.1.12-28).
»libxext6:i386« ist bereits die neuste Version (2:1.3.3-1).
»libxi6:i386« ist bereits die neuste Version (2:1.7.6-1).
»libxrender1:i386« ist bereits die neuste Version (1:0.9.9-0ubuntu1).
»libxtst6:i386« ist bereits die neuste Version (2:1.2.2-1).
»libc6-i386« ist bereits die neuste Version (2.23-0ubuntu5).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 42 nicht aktualisiert.
/etc/udev/rules.d/70-lego.rules found.
groupadd: Gruppe »lego« existiert bereits.
Benutzer steffen wird zur Gruppe lego hinzugefügt.
Der Benutzer »steffen« ist bereits ein Mitglied der Gruppe »dialout«._

_steffen@lenovix:~$ id
uid=1000(steffen) gid=100(users) Gruppen=100(users),4(adm),20(dialout),24(cdrom),27(sudo),30(dip),46(plugdev),113(lpadmin),128(sambashare),1003(lego)_

Further, would be nice when the installer copies the file to a shared folder e.g. /usr/share/OpenRobertaUSB, because usually you download to ~/Download, extract there and execute install.sh with expectation it'll install it to another location then.

cvejoski, 2017-02-07:
From what I can see is that you are running the jar with your default java installation which is 64 bit.

Try
steffen@lenovix:/usr/share/OpenRobertaUSB/java/bin$ ./java -jar -Dfile.encoding=utf-8 ../../OpenRobertaUSB.jar

instead of 

steffen@lenovix:/usr/share/OpenRobertaUSB/java/bin$ java -jar -Dfile.encoding=utf-8 ../../OpenRobertaUSB.jar

Then you will use the java that is provided with the installer, also you can find **Open Roberta USB** application in the application menu.

Yes, that actually nice idea to copy the files to some other location

tuxlego, 2017-02-08:
sorry - still too much Windows in my head ;-)
It works, great job! Thanks supporting the Tux community!

closed 2017-02-11 by bjost2s