OpenRoberta/robertalab #395: The NEPO debugger (Medium/Hard)

cvejoski, 2017-02-09:
All modern editors have possibility of debugging your code. In our Open Roberta® Lab we would like to have the option of debugging a program written in the NEPO language for robots system (EV3) that have real time connection with the Open Roberta Lab. In the first step of this project the student would design a system that allows displaying on the client the current executed block as well as the readings from all the connected sensors. For the second step (optionally) integrated more traditional options for a debugger (controlling the execution of the program, evaluating the variables.)

On the technical side, the project involves:

- adding UI support for highlighting the current executed code
- adding UI support for showing the sensor values

Mentor

Krashcan, 2017-02-23:
Can I work on this? I feel I can do this with a little help.

rbudde, 2019-06-26:
in the wiki

closed 2019-06-26 by rbudde