OpenRoberta/robertalab #396: Questions about local open roberta lab server.

DJuego, 2017-02-10:
I am using the local open roberta lab server according the README.md instructions (./ora.sh --start-from-git)  It works!

However,  I do not know how to stop the server in a safe way. I have to use a rough Ctrl+C

Also I can not access the users of the previous session when I launch the server again. Is the database reset to empty database? 

Is this normal? 🔢

bjost2s, 2017-02-11:
Yes and no,
there is no other way to stop the server than killing it. If you run it in the background eg with nohup you have to search for the process and kill it (`ps aux|grep java`).
The instructions in the readme are for developers. We have to update our wiki, because for starting the server in a productive environment we suggest to use the export parameter and run the database and the server seperately. Than your database is save.

DJuego, 2017-02-11:
Thank you for your swift answer, @bjost2s  
Probably I should have stated that I am using windows. :-)

Please, could we have the step-by-step instructions to launch the server in a productive environment? I need the database! Thanks!

bjost2s, 2017-02-11:
I could try it by heart. First of all I always use the git bash when working with the server on windows.
- From robertalab/OpenRobertaParent just do a normal `mvn clean install` or for real productive environment `mvn clean install -Prelease` to reduce loading time.
- go back to the folder robertalab
- use `./ora.sh --export /path/andFolder/fromWhereYouWantToRunThe/Server/`, don't miss the last /
- go to the new folder and create inside a new folder db `mkdir db`
- If you already have a database, that you would like to use, please copy the content of the former db folder into the new one
- If you would like to have a new, fresh and empty database run in robertalab `./ora.sh --createemptydb  /path/andFolder/fromWhereYouWantToRunThe/Server/db`
- In any case in your new db folder there should be at least two files with names starting with openroberta-db. . Please check this.
- Go back to the new server folder with `cd ..`
- Start the database with `nohup ./start-db.sh &`
- wait a minute and start the server with `./start-server.sh -d server.port=1999 -d robot.whitelist=sim,ev3,calliope,nxt,microbit,ardu,nao -d plugin.tempdir=/tmp/`

With the -d flag you can overwrite the default configurations. The port is important to know how to access later the server. With the robot whitelist you can easily reduce the number of supported robots. The plugin.tempdir is in this example for a linux system. Please check if this is ok for windows. The folder will take the user generated sources and compiled programs and can become huge if you work with ev3's.

If everything works fine, you can also surround the server start with nohup and & to run it in the background.

After you restarted your computer you always have to restart the database first and then the server.

bjost2s, 2017-02-11:
Sometimes one run into right issues. Please check if you have full write access to the new copied folder. For this you can try `sudo chmod -R 777 yourNewFolder`

DJuego, 2017-02-11:
Hi! Thank you for this extensive explanation @bjost2s 

I have tried to reproduce the procedure. The two files in the new db folder are:

db.properties
db.script

When i run: `noup ./start-db.sh` i got

> 
> > nohup ./start-db.sh &
> nohup: ignoring input and appending output to 'nohup.out'

I have to do CTRL+C for recovering the control. 

In the nohup.out i got:

> 
> [Server@1f32e575]: [Thread[main,5,main]]: checkRunning(false) entered
> [Server@1f32e575]: [Thread[main,5,main]]: checkRunning(false) exited
> [Server@1f32e575]: Startup sequence initiated from main() method
> [Server@1f32e575]: Could not load properties from file
> [Server@1f32e575]: Using cli/default properties only
> [Server@1f32e575]: Initiating startup sequence...
> [Server@1f32e575]: Server socket opened successfully in 15 ms.
> [Server@1f32e575]: Database [index=0, id=0, db=file:db/openroberta-db, alias=openroberta-db] opened sucessfully in 189 ms.
> [Server@1f32e575]: Startup sequence completed in 204 ms.
> [Server@1f32e575]: 2017-02-11 12:55:44.793 HSQLDB server 2.3.2 is online on port 9001
> [Server@1f32e575]: To close normally, connect and execute SHUTDOWN SQL
> [Server@1f32e575]: From command line, use [Ctrl]+[C] to abort abruptly

I had tried to open a new console and 👍 

`sh ./start-server.sh -d server.port=1999 -d robot.whitelist=sim,ev3,calliope,nxt,microbit,ardu,nao -d plugin.tempdir=/tmp/`

> 
> 2017-02-11 13:12:07,187 [main] session-id=[] user-id=[] robot-name=[] INFO org.hibernate.cfg.Configuration - HHH000043: Configuring from resource: hibernate-cfg.xml
> 2017-02-11 13:12:07,188 [main] session-id=[] user-id=[] robot-name=[] INFO org.hibernate.cfg.Configuration - HHH000040: Configuration resource: hibernate-cfg.xml
> 2017-02-11 13:12:07,212 [main] session-id=[] user-id=[] robot-name=[] INFO org.hibernate.cfg.Configuration - HHH000041: Configured SessionFactory: null
> 2017-02-11 13:12:07,234 [main] session-id=[] user-id=[] robot-name=[] INFO o.h.s.j.c.i.ConnectionProviderInitiator - HHH000130: Instantiating explicit connection provider: org.hibernate.service.jdbc.connections.internal.C3P0ConnectionProvider
> 2017-02-11 13:12:07,235 [main] session-id=[] user-id=[] robot-name=[] INFO o.h.s.j.c.i.C3P0ConnectionProvider - HHH010002: C3P0 using driver: org.hsqldb.jdbcDriver at URL: jdbc:hsqldb:hsql://localhost/openroberta-db
> 2017-02-11 13:12:07,235 [main] session-id=[] user-id=[] robot-name=[] INFO o.h.s.j.c.i.C3P0ConnectionProvider - HHH000046: Connection properties: {user=orA, password=****, autocommit=false}
> 2017-02-11 13:12:07,235 [main] session-id=[] user-id=[] robot-name=[] INFO o.h.s.j.c.i.C3P0ConnectionProvider - HHH000006: Autocommit mode: false
> feb 11, 2017 1:12:07 PM com.mchange.v2.log.MLog <clinit>
> INFO: MLog clients using java 1.4+ standard logging.
> feb 11, 2017 1:12:07 PM com.mchange.v2.c3p0.C3P0Registry banner
> INFO: Initializing c3p0-0.9.2.1 [built 20-March-2013 10:47:27 +0000; debug? true; trace: 10]
> feb 11, 2017 1:12:07 PM com.mchange.v2.c3p0.impl.AbstractPoolBackedDataSource getPoolManager
> INFO: Initializing c3p0 pool... com.mchange.v2.c3p0.PoolBackedDataSource@bda8ba51 [ connectionPoolDataSource -> com.mchange.v2.c3p0.WrapperConnectionPoolDataSource@9f9b26d2 [ acquireIncrement -> 3, acquireRetryAttempts -> 30, acquireRetryDelay -> 1000, autoCommitOnClose -> false, automaticTestTable -> null, breakAfterAcquireFailure -> false, checkoutTimeout -> 0, connectionCustomizerClassName -> null, connectionTesterClassName -> com.mchange.v2.c3p0.impl.DefaultConnectionTester, debugUnreturnedConnectionStackTraces -> false, factoryClassLocation -> null, forceIgnoreUnresolvedTransactions -> false, identityToken -> 1hge1419mckict3113212i|59252cb6, idleConnectionTestPeriod -> 3600, initialPoolSize -> 5, maxAdministrativeTaskTime -> 0, maxConnectionAge -> 0, maxIdleTime -> 7200, maxIdleTimeExcessConnections -> 0, maxPoolSize -> 100, maxStatements -> 100, maxStatementsPerConnection -> 0, minPoolSize -> 5, nestedDataSource -> com.mchange.v2.c3p0.DriverManagerDataSource@7a4de92c [ description -> null, driverClass -> null, factoryClassLocation -> null, identityToken -> 1hge1419mckict3113212i|79d94571, jdbcUrl -> jdbc:hsqldb:hsql://localhost/openroberta-db, properties -> {user=******, password=******, autocommit=false} ], preferredTestQuery -> null, propertyCycle -> 0, statementCacheNumDeferredCloseThreads -> 0, testConnectionOnCheckin -> false, testConnectionOnCheckout -> false, unreturnedConnectionTimeout -> 0, usesTraditionalReflectiveProxies -> false; userOverrides: {} ], dataSourceName -> null, factoryClassLocation -> null, identityToken -> 1hge1419mckict3113212i|1b765a2c, numHelperThreads -> 3 ]
> feb 11, 2017 1:12:40 PM com.mchange.v2.resourcepool.BasicResourcePool$ScatteredAcquireTask run
> WARNING: com.mchange.v2.resourcepool.BasicResourcePool$ScatteredAcquireTask@2103bac9 -- Acquisition Attempt Failed!!! Clearing pending acquires. While trying to acquire a needed new resource, we failed to succeed more than the maximum number of allowed acquisition attempts (30). Last acquisition attempt exception:
> java.sql.SQLInvalidAuthorizationSpecException: invalid authorization specification - not found: orA
>         at org.hsqldb.jdbc.JDBCUtil.sqlException(Unknown Source)
>         at org.hsqldb.jdbc.JDBCUtil.sqlException(Unknown Source)
>         at org.hsqldb.jdbc.JDBCConnection.<init>(Unknown Source)
>         at org.hsqldb.jdbc.JDBCDriver.getConnection(Unknown Source)
[... cut ...]
>         at com.mchange.v2.resourcepool.BasicResourcePool.awaitAvailable(BasicResourcePool.java:1418)
>         at com.mchange.v2.resourcepool.BasicResourcePool.prelimCheckoutResource(BasicResourcePool.java:606)
>         at com.mchange.v2.resourcepool.BasicResourcePool.checkoutResource(BasicResourcePool.java:526)
>         at com.mchange.v2.c3p0.impl.C3P0PooledConnectionPool.checkoutAndMarkConnectionInUse(C3P0PooledConnectionPool.java:755)
>         at com.mchange.v2.c3p0.impl.C3P0PooledConnectionPool.checkoutPooledConnection(C3P0PooledConnectionPool.java:682)
>         ... 52 more
>

bjost2s, 2017-02-11:
Could you please make sure that the two files in db are named openroberta-db.script and openroberta-db.properties and start the db again?

DJuego, 2017-02-11:
My fault. It works! 👍 Thanks again!
I can restart the database and the server, and my user (and programs) are preserved! Fantastic!

However 

When i run: noup ./start-db.sh i got

nohup ./start-db.sh &
nohup: ignoring input and appending output to 'nohup.out'
I have to do CTRL+C for recovering the control.

is it normal? It should run in the background. True?

I have to open a new console and:

sh ./start-server.sh -d server.port=1999 -d robot.whitelist=sim,ev3,calliope,nxt,microbit,ardu,nao -d plugin.tempdir=/tmp/

It works. But i am not sure it it is what is supposed to be done. (???)

ensonic, 2017-02-19:
We really should make use of the startup script I added. I'll rewrite it to start a db.

closed 2017-11-25 by bjost2s