OpenRoberta/robertalab #408: Variable Handling?

ScottMcl, 2017-02-21:
Wondering if the variable "item" in the attached program is being handled correctly.

Thanks.
[TextVariable.zip](https://github.com/OpenRoberta/robertalab/files/791356/TextVariable.zip)

ScottMcl, 2017-02-21:
Oh.... I see a post below mine about the simulator being unable to break out of a loop. Perhaps this is why my program doesn't stop.

ScottMcl, 2017-02-21:
I'll test it with a brick later today.

ScottMcl, 2017-02-21:
Yes, the program fails to break out of the loop with the simulator but works fine with the brick.

cvejoski, 2017-02-22:
We have bug in the break statement for the simulation. Currently we work on the bug and for the next release probably it will be fixed.

alesolano, 2017-02-23:
Same problem as #406 

Should I close the other issue?

cvejoski, 2017-02-24:
Yes you can close it.

closed 2017-03-16 by cvejoski