OpenRoberta/robertalab #410: New blocks for calliope / microbit

bjost2s, 2017-02-22:
- get Pixel (led from matrix)
- set Pixel
- support for groove connectors

larspelz, 2017-02-28:
I hope it's not too late to say...
I'd really like to see a more low-level access to the accelerometer, something like the PXT has:
`getOrientationAngle({x,y,z})`
`getAcceleration({x,y,z})`
In beginner mode, `getAcceleration()` would be nice, giving just the resultant of the three vectors.
Should I create a new issue for this idea?

cvejoski, 2017-02-28:
No, I think we modify this ticket

larspelz, 2017-02-28:
Cool, thanks a lot!

bjost2s, 2017-02-28:
getAcceleration() seems to be similar to the gesture event 3g,6g,8g (also not implemented yet ;-) but maybe it is easier for kids to understand the function of this sensor in general?

larspelz, 2017-03-01:
`getAcceleration()` should deliver a number, whereas the gesture events deliver boolean values as far as I can see.
The way `gesture ... active?` is implemented neatly gives students a basic understanding of what can be achieved with the sensor. But for more complicated projects, e.g. a bicycle acceleration meter or water rocket flight timer (very enthusiastic, but feasible!), access to the sensor's raw readings would be appreciated I think.

closed 2017-03-30 by cvejoski