OpenRoberta/robertalab #412: Please provide IRC or gitter info

Krashcan, 2017-02-23:
There is no way a student can contact the developer-team to get started with contributing . Please provide gitter or irc where the discussions can take place.

cvejoski, 2017-02-24:
you can join the irc channel **open-roberta**

bjost2s, 2017-02-24:
We offer several ways of communication. Please read 
https://github.com/OpenRoberta/robertalab/wiki/Community

Krashcan, 2017-02-25:
I went to the irc channel. There weren't many people active there. I wish to start contributing to the project and be a part of GSoC'17 if we do get selected. I have setup the project locally. Any suggestion on which issue I should start from?

cvejoski, 2017-02-25:
During the weekends there are not many people active on the IRC channel. Regarding on which issue to start working that depends on you, any preferences client or server side?

Krashcan, 2017-02-25:
I prefer server side

Krashcan, 2017-02-26:
I can try to solve both , client and server side issues.

closed 2017-03-03 by cvejoski