OpenRoberta/robertalab #413: for loop cannot be used to count down

larspelz, 2017-02-24:
In a for loop (expert blocks), the "<" operator will always be used in the condition statement, even if the start count is larger than the end count. Thus, for loops that count down will never execute their loop body, as their condition always yields false.
I created an example to show the problem for the Calliope platform, but I noticed that the problem will persist when using other platforms, too.
[nepo-calliope-for-loop-no-downcount.zip](https://github.com/OpenRoberta/robertalab/files/800428/nepo-calliope-for-loop-no-downcount.zip)

Krashcan, 2017-02-26:
Can I work on this issue?

bjost2s, 2017-02-27:
Yes please! Please send a pull request against develop, if you have a solution or maybe we can have a look at it in your fork and give you some feedback

bjost2s, 2017-02-27:
Feel free to ask, if you need help

Krashcan, 2017-02-27:
I am trying to understand the source code. Can you direct me once? I am looking for the folder where we have code for generating source code from blocks. I might find it if I continue looking around but it is my first time contributing to such a huge project and I am kind of lost. Any hint would help.

Krashcan, 2017-02-27:
Okay, I got it. Its under robertalab\OpenRobertaRobot\src\main\java\de\fhg\iais\roberta\syntax\codegen . Working on the issue.

Krashcan, 2017-02-27:
I have made my changes locally. Do I need to re-run mvn clean install? But it takes time, what should I do next to check if my changes work as expected locally?

cvejoski, 2017-02-27:
If you run the server using the _ora.sh_ script you should build the project again. You can use _mvn clean install -DskipTests_ and it will be faster since the test won't be executed. 

If you run the server using eclipse then you don't have to build the project.

You can create program in Blockly and see the generated source code.

Krashcan, 2017-02-27:
I have made a pull request.

bjost2s, 2017-03-01:
Thank you. Maybe we could improve this and provide a more generic solution. The current solution is only working, if the summand is not an expression or a function. We could have a look in how blockly is doing it for variables eg.
[image]
https://blockly-demo.appspot.com/static/demos/code/index.html

commit 428361a7 (2017-02-28): fixes #413

Uses ">" when there is a negative counter, "<" otherwise.
commit 42e6770a (2017-02-28): Merge pull request #421 from Krashcan/develop

fixes #413
closed 2017-03-30 by cvejoski