OpenRoberta/robertalab #415: drawPicture is completely broken on ev3dev

bidaian, 2017-02-25:
The code generated for drawPicture int NEPOprog.py is something like this:

`hal.drawPicture(u'\u0000\u0000...\u00ff\u0003\u0000\u0000\u00c0\u00ff\u0000\u0000\u00fc\u00ff\u00ff\u00ff\u000f\u0000\u0000\u0000\u00fc\u0007\u0000\u00c0\u00ff\u00ff\u00ff\u00ff\u0003\u0000\u0000\u00c0\u00ff\u0000\u0080\u00ff\u00ff\u00ff\u00ff\u001f\u0000\u0000\u0000\u00fe\u0003\u0000\u00f0\u00ff\u00ff...\u0000', 0, 0)`

Then, in roberta/ev3.py the function does this:

```
def drawPicture(self, picture, x, y):
        if picture not in self.images:
            self.images[picture] = Image.frombytes('1', (178, 128),
                                                   IMAGES[picture],
                                                   'raw', '1;IR', 0, 1)
        self.lcd.image.paste(self.images[picture], (x, y))
        self.lcd.update()
```
The first parameter is used as a key (both in self.images and IMAGES) but nobody seems to loading IMAGES. And the call sends the image data as the first parameter! (the reason why it fails in the first place).

I tried creating the entry it before calling hal.drawPicture, or changing roberta/ev3.py to anonymously create and draw the image from the string, but I get this error:

`TypeError: 'str' does not support the buffer interface`

It's related to a change between python2 and python3. I don't know how to solve it. If you know how, then I'm ready to test it.

Find attached the NEPOprog.xml and the NEPOprog.py generated.  I hope it helps.

[nepoprog.zip](https://github.com/OpenRoberta/robertalab/files/800947/nepoprog.zip)

ensonic, 2017-02-28:
You seem to look at the wrong code:
https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/roberta/ev3.py#L155

I think your ev3dev system is not update to date.

bidaian, 2017-03-01:
Weird ... I was looking at the code in the robot filesystem ... and it listed version as 1.7.1 (and the function is right in the deb package at the repository). Not sure what happened there, but for sure it's not a robertalab issue. Thanks!

closed 2017-03-01 by bidaian