OpenRoberta/robertalab #416: Problem setting up server in local environment

Krashcan, 2017-02-25:
I am on a windows 10 pc and have followed the instructions mentioned in the readme. It works fine up until the last part `./ora.sh --start-from-git`. Everytime I do this, I get

    $ ./ora.sh --start-from-git
    2017-02-25 18:50:37,693 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: openRobertaMailServer.properties
    2017-02-25 18:50:37,693 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - default properties from classpath. Using the resource: openRoberta.properties
    2017-02-25 18:50:37,693 [main] session-id=[] user-id=[] robot-name=[] INFO d.f.iais.roberta.main.ServerStarter - new property from command line: hibernate.connection.url=jdbc:hsqldb:file:OpenRobertaServer/db/openroberta-db
    2017-02-25 18:50:37,693 [main] session-id=[] user-id=[] robot-name=[] INFO d.f.i.roberta.util.RobertaProperties - As temporary directory C:\Users\KUMARA~1\AppData\Local\Temp\ will be used
    2017-02-25 18:50:37,708 [main] session-id=[] user-id=[] robot-name=[] INFO org.eclipse.jetty.util.log - Logging initialized @558ms
    2017-02-25 18:50:37,739 [main] session-id=[] user-id=[] robot-name=[] INFO d.f.i.r.r.RobotCommunicator - timer thread created
    2017-02-25 18:50:37,739 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: Robot.properties
    2017-02-25 18:50:37,777 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: EV3.properties
    2017-02-25 18:50:37,793 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: Robot.properties
    2017-02-25 18:50:37,793 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: Calliope.properties
    2017-02-25 18:50:37,808 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: Robot.properties
    2017-02-25 18:50:37,808 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: NXT.properties
    2017-02-25 18:50:37,808 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: Robot.properties
    2017-02-25 18:50:37,808 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: Microbit.properties
    2017-02-25 18:50:37,824 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: Robot.properties
    2017-02-25 18:50:37,824 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: Ardu.properties
    2017-02-25 18:50:37,824 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: Robot.properties
    2017-02-25 18:50:37,824 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - properties from classpath. Using the resource: NAO.properties
    2017-02-25 18:50:37,840 [main] session-id=[] user-id=[] robot-name=[] INFO d.f.iais.roberta.main.ServerStarter - ROBOT PLUGINS: 6 plugins are found: sim ev3 calliope nxt microbit ardu nao
    2017-02-25 18:50:37,877 [main] session-id=[] user-id=[] robot-name=[] WARN o.e.j.server.handler.ContextHandler - o.e.j.s.ServletContextHandler@49c43f4e{/,null,null} contextPath ends with /*
    2017-02-25 18:50:37,877 [main] session-id=[] user-id=[] robot-name=[] WARN o.e.j.server.handler.ContextHandler - Empty contextPath
    2017-02-25 18:50:37,893 [main] session-id=[] user-id=[] robot-name=[] INFO org.eclipse.jetty.server.Server - jetty-9.2.17.v20160517
    Feb 25, 2017 6:50:38 PM com.sun.jersey.api.core.PackagesResourceConfig init
    INFO: Scanning for root resource and provider classes in the packages:
      de.fhg.iais.roberta.javaServer.restServices.all
      de.fhg.iais.roberta.javaServer.restServices.robot
      de.fhg.iais.roberta.javaServer.provider
    Feb 25, 2017 6:50:38 PM com.sun.jersey.api.core.ScanningResourceConfig logClasses
    INFO: Root resource classes found:
      class de.fhg.iais.roberta.javaServer.restServices.robot.Ev3Update
      class de.fhg.iais.roberta.javaServer.restServices.robot.RobotSensorLogging
      class de.fhg.iais.roberta.javaServer.restServices.all.ClientToolbox
      class de.fhg.iais.roberta.javaServer.restServices.all.Alive
      class de.fhg.iais.roberta.javaServer.restServices.all.ClientPing
      class de.fhg.iais.roberta.javaServer.restServices.all.ClientUser
      class de.fhg.iais.roberta.javaServer.restServices.robot.RobotDownloadProgram
      class de.fhg.iais.roberta.javaServer.restServices.all.ClientAdmin
      class de.fhg.iais.roberta.javaServer.restServices.all.ClientConfiguration
      class de.fhg.iais.roberta.javaServer.restServices.all.ClientProgram
      class de.fhg.iais.roberta.javaServer.restServices.robot.RobotCommand
      class de.fhg.iais.roberta.javaServer.restServices.all.RestExample
    Feb 25, 2017 6:50:38 PM com.sun.jersey.api.core.ScanningResourceConfig logClasses
    INFO: Provider classes found:
      class de.fhg.iais.roberta.javaServer.provider.JAXBMarshallerProvider
      class de.fhg.iais.roberta.javaServer.provider.JAXBContextProvider
      class de.fhg.iais.roberta.javaServer.provider.OraDataProvider
      class de.fhg.iais.roberta.javaServer.provider.JAXBUnmarshallerProvider
    Feb 25, 2017 6:50:38 PM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
    INFO: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientAdmin as a root resource class
    Feb 25, 2017 6:50:38 PM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
    INFO: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientConfiguration as a root resource class
    Feb 25, 2017 6:50:38 PM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
    INFO: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientToolbox as a root resource class
    Feb 25, 2017 6:50:38 PM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
    INFO: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientProgram as a root resource class
    Feb 25, 2017 6:50:38 PM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
    INFO: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientUser as a root resource class
    Feb 25, 2017 6:50:38 PM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
    INFO: Registering de.fhg.iais.roberta.javaServer.restServices.robot.RobotDownloadProgram as a root resource class
    Feb 25, 2017 6:50:38 PM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
    INFO: Registering de.fhg.iais.roberta.javaServer.restServices.robot.RobotCommand as a root resource class
    Feb 25, 2017 6:50:38 PM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
    INFO: Registering de.fhg.iais.roberta.javaServer.restServices.robot.RobotSensorLogging as a root resource class
    Feb 25, 2017 6:50:38 PM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
    INFO: Registering de.fhg.iais.roberta.javaServer.restServices.all.RestExample as a root resource class
    Feb 25, 2017 6:50:38 PM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
    INFO: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientPing as a root resource class
[... 81 lines cut ...]
            at com.mchange.v2.async.ThreadPoolAsynchronousRunner$PoolThread.run(ThreadPoolAsynchronousRunner.java:648) ~[mchange-commons-java-0.2.3.4.jar:na]
    Caused by: org.hsqldb.HsqlException: user lacks privilege or object not found: ROBOT
            at org.hsqldb.error.Error.error(Unknown Source) ~[hsqldb-2.3.2.jar:2.3.2]
            at org.hsqldb.error.Error.error(Unknown Source) ~[hsqldb-2.3.2.jar:2.3.2]
            at org.hsqldb.SchemaManager.getTable(Unknown Source) ~[hsqldb-2.3.2.jar:2.3.2]
[... cut ...]
            at org.hsqldb.StatementManager.compile(Unknown Source) ~[hsqldb-2.3.2.jar:2.3.2]
            at org.hsqldb.Session.execute(Unknown Source) ~[hsqldb-2.3.2.jar:2.3.2]
            ... 8 common frames omitted

I can see it says, couldn't connect to database but I have created a folder db in in server folder and ran the `./ora.sh --createemptydb OpenRoberta/db/openroberta-db` command. After this didn't work, I even tried simply copying dbbase under the name db but the output was same. What should I do?

cvejoski, 2017-02-25:
Where is the location of the *db* folder? By default It should be in the OpenRobertaServer if you do not want to change the path.

Krashcan, 2017-02-25:
It is in the OpenRobertaServer folder.

cvejoski, 2017-02-25:
Can you please clone again the repository, copy dbBase folder as db and run mvn clean install.
This should work I have just tried on Win 10.

Krashcan, 2017-02-25:
It worked after I re-did the whole thing. I must have done some mistake the first time. Also, I noticed, in the readme to create a new empty database, the command give is `./ora.sh --createemptydb OpenRoberta/db/openroberta-db` , Shouldn't it be `./ora.sh --createemptydb OpenRobertaServer/db/openroberta-db` ?

cvejoski, 2017-02-25:
Yes, that is correct the read me file is not update

cvejoski, 2017-02-27:
fixed with 35be632

closed 2017-02-27 by cvejoski