OpenRoberta/robertalab #417: Calliope: erratic random number generation on variable initialisation

larspelz, 2017-02-26:
If the random generator is used to initialise a variable, it always yields the minimum value
on the Calliope hardware. In the simulation, it works as expected.
Please try the example given below.
[image]
I attached a demo program for convenience:
[nepo-calliope-random-generator-erratic-behaviour.zip](https://github.com/OpenRoberta/robertalab/files/802109/nepo-calliope-random-generator-erratic-behaviour.zip)

bjost2s, 2017-02-27:
This is because at the moment of the variable declaration, `ubit.init()` is not executed. Please set rnd to the random value in the beginning of your program like in this example.
[image]

larspelz, 2017-02-28:
Thank you for the advice! As programmer, I can easily adapt to the given solution. As a teacher, I am hard pressed to explain this "workaround", since the possibility of initialising the variable with a random number like I did in the first example seems more obvious to me. If students do it like this, they cannot easily devise their own solution, since neither syntactical nor semantical errors are reported.
Is it possible for the code generator to automatically generate the code you gave from the representation in the first post? Maybe if a check is implemented to see whether a random number is used to _initialise_ a variable?

bjost2s, 2017-02-28:
I think this is hard to implement, because of the variety of languages we already have implemented. What we could do easily is to make the declaration more save that users cannot change the preassigned blocks, but changing the value. The user than cannot assign a function to the declaration anymore.

larspelz, 2017-03-01:
That at least would be a good idea. Today, I presented the editor to students who used to work with Scratch. They immediately put the random number generator into the initialisation. But for now, an error message or so would be good.

elhele, 2017-07-20:
Fixed it in issue  #483 with 3c52e2c commit

closed 2017-09-11 by cvejoski