OpenRoberta/robertalab #425: Add a speak/say block

ensonic, 2017-03-03:
On ev3dev we have espeak to produce synthetic speech from any text. I also managed to build this for lejos and if you search for espeak.js you find many solution where espeak was compiled to be used in the browser.

The only robot that can speak already is the NAO. There we have two blocks:
* [ set language <language> ]
* [ say ]-[ \<text\> ]

Espeak supports over 60 languages. Interesting parameters that could be exposed in the block are:

-a <integer> Amplitude, 0 to 200, default is 100
-p <integer> Pitch adjustment, 0 to 99, default is 50
-s <integer> Speed in words per minute, 80 to 450, default is 175

The amplitude we can take from the [ set volume ] block.

We can introduce a stateful [ set language ] block for ev3 as well, or use the program ui language. Any opinions?

JanisM91, 2017-03-03:
There also is a third Block that is related to synthetic speech for NAO:
* [ get language]

This Block returns the language the speech engine of the robot is currently set to.

ensonic, 2017-03-03:
@JanisM91 what is [ get language ] good for? Also does the speech synth on the NAO have features to control pitch and speed?

JanisM91, 2017-03-03:
@ensonic The user might want to know which language the speech synth is currently set to. The language is not initialized. therefore the user might not know the current language and just change it if it is not the desired language. Another possible use of this block is very specific for the NAO. The user might want to develop a programm that can be used with NAO in different countries. The user could use the [ get language ] block to identify the language and send text in the correct language to the speech synth. This need only arises on the NAO because every NAO has one default language (usually english) and one language that the user can download. You have to pay for additional languages. So most NAOs will only support two languages.

It is possible to control pitch and speed. The blocks currently don't support this. But I think it is a good idea to add this option.

ensonic, 2017-09-08:
We'll consider to add [ set language ] blocks to all robots that will receive the [ say ] block. The advantage here is that this lets us change the language during the program as well. So the [ set language ] will affect a  program wide global state.

boonto, 2017-12-07:
Currently the EV3 implementations reuse the naoActions_sayText & naoActions_setLanguage blockly blocks. In the latter block the drop down menu languages are specified.
I will work on fixing this next.
Then I will move on to #722

ensonic, 2017-12-07:
We should specify what defaults we want to use for all robots that use espeak. I had those:
https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/roberta/ev3.py#L323
But I'll need to make another release to add the pitch and speed params (will do this next week). Together with that I'll also adjust the defaults in my impl to what got submitted for lejos.

closed 2018-01-30 by bjost2s