OpenRoberta/robertalab #434: NAO - Charset for String Blocks

fgras-erm, 2017-03-08:
Hi,

When I entry a text with accent _(é è ç à ï ...)_ in String Bock, the generated python code replace the text with a empty string

Regards,
fgras-erm

JanisM91, 2017-03-09:
I tried to entry a small letter a with grave into the String block. For me it is not replaced by an empy string but with the corresponding unicode.

[image]

fgras-erm, 2017-03-10:
Hi,

Indeed, the first time I tested, I had the same thing than you.
The second test, I had nothing, maybe it's because I wasn't on the good version or on good server.

Nao works on Python 2.7.  My version, on Windows, doesn't support "\u00E0" _(I'm out of NAO this week, but I rather am sure it's the same on and in NAO)_
NAOqi 2.1.4 doesn't support unicode. Consequently, the communication with NAO must be in str in utf8, it should be either 
`self.tts.say("\xC3\cA0")`
or
`self.tts.say( u"à".encode('utf8') )`

Regards
fgras

JanisM91, 2017-08-01:
The special characters in the strings are converted on the server. Therefore the unicode the server generates is converted to a proper utf-8 string in the hal.

closed 2017-09-11 by cvejoski