OpenRoberta/robertalab #435: NAO - Compiler crashes with "Wait Until Block"

fgras-erm, 2017-03-08:
Hi,

When I put a "Wait Until Block" in my program, I can't refresh my generated python code.

In my Firefox console, I can see server answer : 
```
{"cmd":"showSourceP","rc":"error","message":"ORA_COMPILERWORKFLOW_ERROR_PROGRAM_GENERATION_FAILED","cause"
:"ORA_COMPILERWORKFLOW_ERROR_PROGRAM_GENERATION_FAILED","serverTime":"Wed Mar 08 14:43:36 CET 2017","server
.version":"2.2.0"}
```

Regards
fgras-erm

JanisM91, 2017-03-10:
fixed with 7350eaa and 36205c9. will be part of the next release.

[image]

closed 2017-03-10 by JanisM91

fgras-erm, 2017-03-11:
Hi,

Caution time.sleep is in second. It seems the program will perform one test every 15 seconds. I thinks you want to use millisecond? 

What is the two Exception for?

Regards,
fgras

JanisM91, 2017-03-11:
Hi,

it is correct this should be milliseconds. I will open a new ticket for this. Solution will be a new method in the hal and the generated code will use this method. For EV3 it is done in the same way.

The exceptions are part of the solution for #328 .